stevebuk1/pv_opt #391: AttributeError: 'NoneType' object has no attribute 'index' in pv_opt error.log

swests, 2025-02-18:
**Describe the bug**
Seeing this error (in riemann_avg) periodically in the error.log

HA: 2025:02:04
PV Opt: v4.0.8

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

stevebuk1, 2025-02-18:
This is most likely to be that the data read for consumption in W is taking longer than 10 seconds, which is an AppDaemon hard limit. 

It is better to read the consumption figures from a kWh sensor, as the amount of data to be transferred is much smaller. Can you post a pv_opt.log so I can check which one you've got setup?

swests, 2025-02-19:
Log from last start-up attached.  WHen I see this again I'll attached the log from when it happens

[pv_opt.log](https://github.com/user-attachments/files/18864182/pv_opt.log)

stevebuk1, 2025-02-19:
As I suspected you are using a power sensor and loading 21 days of history from it. This has been known to cause issues if things are are a little slow (HA hardware peformance, general HA database size, memory card read performance( which will slow with age)). 

Its now recommended to use a kWh sensor instead of a W sensor to avoid these problems. 

v4.0.8 utilises fake values for W consumption sensors so that kWh consumption sensors are used in preference. 

To utilise the kWh consumption sensor, in config.yaml just comment out any entries under the line   `# id_consumption:`. This will mean Pv_opt will default to using the kWh sensor `sensor.solmod_house_load_today` which should then resolve this issue. 

If this doesnt make sense just attach config.yaml and I'll make the edits for you.

stevebuk1, 2025-02-20:
Just realised this is related to a previous issue you raised, #270. 

If you get the same error messages in Pv_opt that you did when you raised the previous issue that will confirm the root cause of this issue, although it sounds like its degraded again from the fix that was put in to increase the cache size.

swests, 2025-02-27:
It's not related to #270 as I now move to previous days window to 7, because I was getting reducing benefit.  I made the change in MQTT but not the config - hence the confusion.

Under id_consumption: I have
    - sensor.{device_name}_house_load
    - sensor.{device_name}_bypass_load
Will comment these out and see what happens.

stevebuk1, 2025-03-01:
Hmm at 7 days it shouldnt be an issue. Lets see what the logs say.

closed 2025-05-09 by github-actions[bot]