stevebuk1/pv_opt #392: ERROR: 145: 'DataFrame' object has no attribute 'dtype'

swests, 2025-02-19:
**Describe the bug**
Attached Error.log and PV_Opt.log from same run (00:10)

[error.log](https://github.com/user-attachments/files/18864253/error.log)

[pv_opt.log](https://github.com/user-attachments/files/18864260/pv_opt.log)

stevebuk1, 2025-02-20:
The 00:10 tried to load the next days IOG rates from the Octopus Energy Integration but there was no data available.

It did detect and handle this correctly here:

`No data found in next day rate`

but then went on to cause an error when trying to update prices with empty data, which aborts the optimiser run. A bugfix is needed.

stevebuk1, 2025-02-22:
Fixed in v4.0.9-Beta7 available as Beta on HACS.

stevebuk1, 2025-03-28:
Released in v4.0.9, closing issue.

closed 2025-03-28 by stevebuk1