stevebuk1/pv_opt #393: Question on Tariff Errors

swests, 2025-02-19:
I see this in the log:

2025-02-19 09:00:00.025898     INFO: Checking tariff prices vs Octopus Energy Integration:
2025-02-19 09:00:00.026466     INFO: -----------------------------------------------------
2025-02-19 09:00:00.026994     INFO: 
2025-02-19 09:00:00.040787     INFO:   Import: Average Prices - PV_OPT: 20.34 p/kWh  OE Integration: 20.34 p/kWh  Mean difference:  0.00 p/kWh
2025-02-19 09:00:00.060952     INFO:   Export: Average Prices - PV_OPT: 12.50 p/kWh  OE Integration: 15.00 p/kWh  Mean difference:  2.50 p/kWh <<< ERROR
2025-02-19 09:00:00.061499     INFO: 
2025-02-19 09:00:00.061984     INFO: Checking tariff start and end times:
2025-02-19 09:00:00.062484     INFO: ------------------------------------
2025-02-19 09:00:00.064635     INFO:   Import: E-1R-INTELLI-VAR-24-10-29-H              Start: 2024-10-29 00:00:00+0000 End: 2025-02-20 05:30:00+0000 
2025-02-19 09:00:00.065208     INFO:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-H         Start: 2019-05-15 23:00:00+0000 End: N/A 
2025-02-19 09:00:00.065720     INFO:   Octopus Intelligent Go tariff detected. Rates will be downloaded at midnight, 04:40 and 16:40, and on car plugin
2025-02-19 09:00:00.066190     INFO:   Tariffs OK

I haven't changed the export tariff in ages, so unsure why it flags

stevebuk1, 2025-02-19:
The error is generated when Pv_opt compares the Octopus Energy Integration information to what it has been told itself and finds a difference. Normally there should be none. 

Its very strange - Pv_opt is reporting 12.5p, Octopus Energy Integration is reporting 15p, but its been 15p for ages, not sure where the 12.5p is coming from. 

In config.yaml, around line 23 please change

`  debug: False`
to
`   debug: True
`
and then add the following line just underneath (making sure the indent matches)

`  debug_categories: OT`

Pv_opt will automatically restart. 

Then post a pv_opt.log please.

swests, 2025-02-27:
Hi Steve

Just made this change.  pv_opt.log attached

[pv_opt.log](https://github.com/user-attachments/files/19006969/pv_opt.log)

swests, 2025-02-27:
Reviewing the log from the 1/2 Hour summary (Lines 520 onwards), looks like pv_opt adjusts down the export price when IOG goes to off-peak rate - see after line 547...

stevebuk1, 2025-02-27:
I think you are reading it correctly. I don't have an export tariff so havent seen this, but have forced one in the latest code and also a version of 3.x.x that I often use for comparative testing, I get the same. I don't think the export tariff of 7.003p during cheap rate is correct but it appears to be a long standing issue and I don't see an impact to it unless you are looking to cyclicly charge and discharge throughout peak rate? 

Not sure if its a bug within Pv_opt or what the Octopus API generates when doing a tariff download from it - I'm inclined to think the former but will investigate.

stevebuk1, 2025-03-01:
Found the issue within Pv_opt. 

v4.0.9-Beta-9 contains the fix, available on HACS as a Beta release. 

Note:  this release contains a bugfix for #395, which was to rename the default option in select.pv_opt_charger from "None" to "Not fitted". Not sure if it effects those without EVs, but it required the deletion of the entity so that Pv_opt can regenerate it with the renamed option. There is no easy way to delete an entity owned by MQTT, but the following method will work:

- Go to Add-Ons and Stop both the Mosquito MQTT broker and Appdaemon
- Go to Services and delete the MQTT integration
- Restart HA
- Start the Mosquito Broker
- Start Appdaemon
- Add the MQTT Integration under Services with the default settings
- Re-start Appdaemon

swests, 2025-03-06:
Hi Steve

Beta installed and yes, the issue is resolved.

Regarding changing default values for select.pv_opt_ev_charger, rather than follow the process described I used the MQTT Explorer add-on to update the pvopt_ev_charger topic manually, changed the config change None to Not fitted. 

[image]

[image]

his felt a little less drastic that removing and re-adding MQTT

closed 2025-03-06 by swests

stevebuk1, 2025-03-06:
Thanks for the advice on dealing with entities. When trying to sort on #280, we knew we couldn't delete entities with MQTT explorer but didn't realize you could update default values, thanks.