stevebuk1/pv_opt #397: Stuck in Loading Tariffs

lazy-pete, 2025-02-27:
**Describe the bug**
A clear and concise description of what the bug is.

Stuck in ‘Loading Tarriffs’ since 1600 on 24/2/25. 

I’ve tried updating the octopus integration, and my dashboard seems to be pulling tariff information. 

Yaml is using octopus auto 

Shall append log files when I get home. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

lazy-pete, 2025-02-27:
Still seems to be stuck. Logs look as if appdaemon not running, though it does seem to be, and I’ve restarted it a couple of times (though it doesn’t seem to start on reboot of HA, which it used to, despite the toggle to do so being on). 

Log files below

[pv_opt.log](https://github.com/user-attachments/files/19015397/pv_opt.log)
[error.log](https://github.com/user-attachments/files/19015398/error.log)
[main.log](https://github.com/user-attachments/files/19015396/main.log)

lazy-pete, 2025-02-27:
Actually - for reasons I don’t understand I don’t seem to have generated any log files since the start of February - can that be true ?

lazy-pete, 2025-03-02:
Got back on to Agile and all is resolved.

closed 2025-03-02 by lazy-pete

stevebuk1, 2025-03-02:
Was about to pick this one up but good to hear its resolved.