stevebuk1/pv_opt #400: Tz_convert errors when loading tariff

swests, 2025-03-29:
Woken up to a stream of errors and PV_Opt stuck at loading tariff. 

I’m Remote from the install but managed to snag this from the error log. Suspect it’s because of the clock change and how octopus are delivering the tariff data.

.

Log Entry
2025-03-29 06:59:50.466426 WARNING pv_opt: ------------------------------------------------------------
Log Entry
Log Entry
AttributeError: 'Index' object has no attribute 'tz_convert'
Log Entry
^^^^^^^^^^^^^^^^^^
Log Entry
y.index = y.index.tz_convert("UTC")
Log Entry
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 823, in get_io_tariffs
Log Entry
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Log Entry
self.io_prices = self.get_io_tariffs(self.octopus_import_entity[0])
Log Entry
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2394, in optimise
Log Entry
^^^^^^^^^^^^^^
Log Entry
return f(*args, **kw)
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
Log Entry
self.optimise()
Log Entry
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 687, in initialize
Log Entry
^^^^^^^^^^^^^^
Log Entry
return f(*args, **kw)
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
Log Entry
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Log Entry
result = self.fn(*self.args, **self.kwargs)
Log Entry
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
Log Entry
^^^^^^^^^^^^^^^
Log Entry
response = future.result()
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
Log Entry
await utils.run_in_executor(self, init)
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
Log Entry
2025-03-29 06:59:50.466342 WARNING pv_opt: Traceback (most recent call last):
Log Entry
2025-03-29 06:59:50.465346 WARNING pv_opt: ------------------------------------------------------------
Log Entry
2025-03-29 06:59:50.465275 WARNING pv_opt: Unexpected error running initialize() for pv_opt
Log Entry
2025-03-29 06:59:50.465181 WARNING pv_opt: ------------------------------------------------------------
Log Entry
2025-03-29 06:56:24.700574 WARNING pv_opt: --------

stevebuk1, 2025-03-29:
Don't worry about logs, as my system is also stuck. Octopus must have
changed something that ripples through to the next_day_rates entity. I'll
investigate this evening.

stevebuk1, 2025-03-29:
Fixed in v4.0.10-Beta-1.

swests, 2025-03-29:
Thank you. Back working again!

swests, 2025-03-30:
Scratch that.  I guess you have the same issue again this morning @stevebuk1   

There are also warnings on the AppDaemon log. 

Log Entry
2025-03-30 07:05:35.729382 WARNING pv_opt: ------------------------------------------------------------
Log Entry
Log Entry
AttributeError: 'Index' object has no attribute 'tz_convert'
Log Entry
^^^^^^^^^^^^^^^^^^
Log Entry
x.index = x.index.tz_convert("UTC")
Log Entry
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 798, in get_io_tariffs
Log Entry
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Log Entry
self.io_prices = self.get_io_tariffs(self.octopus_import_entity[0])
Log Entry
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2391, in optimise
Log Entry
^^^^^^^^^^^^^^
Log Entry
return f(*args, **kw)
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
Log Entry
self.optimise()
Log Entry
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 684, in initialize
Log Entry
^^^^^^^^^^^^^^
Log Entry
return f(*args, **kw)
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
Log Entry
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Log Entry
result = self.fn(*self.args, **self.kwargs)
Log Entry
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
Log Entry
^^^^^^^^^^^^^^^
Log Entry
response = future.result()
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
Log Entry
await utils.run_in_executor(self, init)
Log Entry
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
Log Entry
2025-03-30 07:05:35.729294 WARNING pv_opt: Traceback (most recent call las

stevebuk1, 2025-03-30:
Indeed, it seems there is a  bit more work to do now the list contains all
BST times. I'll look this evening.

stevebuk1, 2025-03-30:
Grr, whilst I fixed next_day_rates, I forgot current_day_rates, so on the roll of midnight the problem transferred from former to latter. 

v4.0.10-Beta-2 released to fix. I'll leave this open for a day or two to ensure I've got them all this time.

swests, 2025-03-30:
Do you also read from  previous ?

Simon West
***@***.***
+44 7774 613073

stevebuk1, 2025-03-30:
> Do you also read from previous ?

Good point, but no. Just current_day and next_day from the bottlecap integration to get the very latest pricing.

swests, 2025-03-31:
That got it squashed. Thanks!

closed 2025-03-31 by swests