stevebuk1/pv_opt #402: New set of timezone issues

swests, 2025-04-22:
Getting the following in the log

```

2025-04-22 12:37:54.384071 WARNING pv_opt: ------------------------------------------------------------
--
 
ValueError: Both dates must have the same UTC offset
raise ValueError("Both dates must have the same UTC offset")
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 6875, in slice_locs
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
start_slice, end_slice = self.slice_locs(start, end, step=step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 6662, in slice_indexer
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return Index.slice_indexer(self, start, end, step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/datetimes.py", line 682, in slice_indexer
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
indexer = labels.slice_indexer(slice_obj.start, slice_obj.stop, slice_obj.step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1443, in _get_slice_axis
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self._get_slice_axis(key, axis=axis)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1411, in _getitem_axis
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self._getitem_axis(maybe_callable, axis=axis)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1191, in __getitem__
~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
return getattr(section, self.name)[new_key]
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1089, in _getitem_lowerdim
^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self._getitem_lowerdim(tup)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1368, in _getitem_tuple
^^^^^^^^^^^^^^^^^^^^^^^^
return self._getitem_tuple(key)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1184, in __getitem__
~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
df.loc[event_start:event_end, "unit"] += event_value
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 385, in to_df
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
imp_df = self.tariffs["import"].to_df(start=start.floor("30min"), end=end, **kwargs)
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 659, in net_cost
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cost_today = self.contract.net_cost(grid_flow=grid, log=self.debug, day_ahead=False, sum=False)
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1424, in _cost_actual
self._cost_actual()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 671, in initialize
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^
response = future.result()
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
2025-04-22 12:37:54.383968 WARNING pv_opt: Traceback (most recent call last):
2025-04-22 12:37:54.381107 WARNING pv_opt: ------------------------------------------------------------
2025-04-22 12:37:54.381040 WARNING pv_opt: Unexpected error running initialize() for pv_opt
2025-04-22 12:37:54.380926 WARNING pv_opt: ------------------------------------------------------------
```

Started at noon today.  I'm guessing Octopus have changed 'something'

swests, 2025-04-22:
More logging (debug options OTFV): Looks like octopus have forgotten it's BST in their data...

```
Log Entry
--
2025-04-22 14:49:32.812472 INFO: >>> Start: 22/04 00:00 UTC End: 22/04 13:30 UTC
2025-04-22 14:49:32.811873 INFO: >>> E-1R-INTELLI-VAR-24-10-29-H
2025-04-22 14:49:32.753318 INFO: >>> Start: 22-Apr 00:00 UTC End: 22-Apr 13:49 UTC
2025-04-22 14:49:32.752806 INFO:
2025-04-22 14:49:32.752321 INFO: No entry found in config.yaml for plug_status sensor, leaving any autodetected sensor in place.
2025-04-22 14:49:32.751821 INFO:
2025-04-22 14:49:32.751265 INFO: sensor.myenergi_zappi_********_plug_status
2025-04-22 14:49:32.750757 INFO: Zappi plug_status sensor(s) found:
2025-04-22 14:49:32.750211 INFO: Attempting to autodetect Zappi Plugin status sensor(s)
2025-04-22 14:49:32.749743 INFO:
2025-04-22 14:49:32.749267 INFO: sensor.myenergi_zappi_********_charge_added_session
2025-04-22 14:49:32.748744 INFO: Zappi consumption sensor(s) found:
2025-04-22 14:49:32.735079 INFO: Attempting to autodetect Zappi consumption sensor(s)
2025-04-22 14:49:32.734631 INFO:
2025-04-22 14:49:32.734154 INFO: ----------------
2025-04-22 14:49:32.733676 INFO: Reading Zappi(s)
2025-04-22 14:49:32.733190 INFO:
2025-04-22 14:49:32.732452 INFO: Finished loading contract
2025-04-22 14:49:32.731844 INFO:
2025-04-22 14:49:32.731236 INFO: 1354: 22-Apr 18:00 UTC+01:00 - 22-Apr 19:00 UTC+01:00 at 11.9p/kWh
2025-04-22 14:49:32.730464 INFO: The following Octopus Saving Events have been joined:
2025-04-22 14:49:32.729753 INFO:
2025-04-22 14:49:32.728086 INFO: Found Octopus Savings Events entity: event.octopus_energy_**********_octoplus_saving_session_events
2025-04-22 14:49:32.727395 INFO:
2025-04-22 14:49:32.722606 INFO:
2025-04-22 14:49:32.721959 INFO: Contract tariffs loaded OK
2025-04-22 14:49:32.721451 INFO:
2025-04-22 14:49:32.720662 INFO: No IOG Smart Charging Schedule found.
2025-04-22 14:49:32.720040 INFO: Octopus Intelligent Go Smart Charging Schedule is....
2025-04-22 14:49:32.719454 INFO:
2025-04-22 14:49:32.718076 INFO: Current Dispatch Status (On/off) = : off
2025-04-22 14:49:32.716832 INFO:
2025-04-22 14:49:32.716285 INFO: -----------------------------------------------------------
2025-04-22 14:49:32.715744 INFO: Reloading IOG Car slots from Intelligent Dispatching Sensor
2025-04-22 14:49:32.715207 INFO:
2025-04-22 14:49:32.714658 INFO: Charge to Add Value = : 80
2025-04-22 14:49:32.713708 INFO: Found Charge to Add entity: number.octopus_energy_**********_intelligent_charge_target
2025-04-22 14:49:32.711102 INFO: Trying to find Car % Charge to add from Octopus Energy Integration
2025-04-22 14:49:32.710573 INFO:
2025-04-22 14:49:32.710026 INFO: Found Dispatching Sensor: binary_sensor.octopus_energy_**********_intelligent_dispatching
2025-04-22 14:49:32.707218 INFO: Trying to find Octopus Intelligent Dispatching Sensor from Octopus Energy Integration
2025-04-22 14:49:32.706706 INFO:
2025-04-22 14:49:32.706160 INFO: -------------------------
2025-04-22 14:49:32.705607 INFO: Getting Car Charging Plan
2025-04-22 14:49:32.705044 INFO:
2025-04-22 14:49:32.623149 INFO: >>>X _load_contract E-1R-OUTGOING-FIX-12M-19-05-13-H
2025-04-22 14:49:32.621228 INFO: >>>export: ['event.octopus_energy_electricity_22j0076737_2000060308364_export_current_day_rates']
2025-04-22 14:49:32.620743 INFO: start 2025-04-21 23:00:00+00:00 7.0003 2025-04-21 23:30:00+00:00 7.0003 2025-04-22 00:00:00+00:00 7.0003 2025-04-22 00:30:00+00:00 7.0003 2025-04-22 01:00:00+00:00 7.0003 2025-04-22 01:30:00+00:00 7.0003 2025-04-22 02:00:00+00:00 7.0003 2025-04-22 02:30:00+00:00 7.0003 2025-04-22 03:00:00+00:00 7.0003 2025-04-22 03:30:00+00:00 7.0003 2025-04-22 04:00:00+00:00 7.0003 2025-04-22 04:30:00+00:00 28.7898 2025-04-22 05:00:00+00:00 28.7898 2025-04-22 05:30:00+00:00 28.7898 2025-04-22 06:00:00+00:00 28.7898 2025-04-22 06:30:00+00:00 28.7898 2025-04-22 07:00:00+00:00 28.7898 2025-04-22 07:30:00+00:00 28.7898 2025-04-22 08:00:00+00:00 28.7898 2025-04-22 08:30:00+00:00 28.7898 2025-04-22 09:00:00+00:00 28.7898 2025-04-22 09:30:00+00:00 28.7898 2025-04-22 10:00:00+00:00 28.7898 2025-04-22 10:30:00+00:00 28.7898 2025-04-22 11:00:00+00:00 28.7898 2025-04-22 11:30:00+00:00 28.7898 2025-04-22 12:00:00+00:00 28.7898 2025-04-22 12:30:00+00:00 28.7898 2025-04-22 13:00:00+00:00 28.7898 2025-04-22 13:30:00+00:00 28.7898 2025-04-22 14:00:00+00:00 28.7898 2025-04-22 14:30:00+00:00 28.7898 2025-04-22 15:00:00+00:00 28.7898 2025-04-22 15:30:00+00:00 28.7898 2025-04-22 16:00:00+00:00 28.7898 2025-04-22 16:30:00+00:00 28.7898 2025-04-22 17:00:00+00:00 28.7898 2025-04-22 17:30:00+00:00 28.7898 2025-04-22 18:00:00+00:00 28.7898 2025-04-22 18:30:00+00:00 28.7898 2025-04-22 19:00:00+00:00 28.7898 2025-04-22 19:30:00+00:00 28.7898 2025-04-22 20:00:00+00:00 28.7898 2025-04-22 20:30:00+00:00 28.7898 2025-04-22 21:00:00+00:00 28.7898 2025-04-22 21:30:00+00:00 28.7898 2025-04-22 22:00:00+00:00 28.7898 2025-04-22 22:30:00+00:00 7.0003 2025-04-22 23:00:00+00:00 7.0003 2025-04-22 23:30:00+00:00 7.0003 2025-04-23 00:00:00+00:00 7.0003 2025-04-23 00:30:00+00:00 7.0003 2025-04-23 01:00:00+00:00 7.0003 2025-04-23 01:30:00+00:00 7.0003 2025-04-23 02:00:00+00:00 7.0003 2025-04-23 02:30:00+00:00 7.0003 2025-04-23 03:00:00+00:00 7.0003 2025-04-23 03:30:00+00:00 7.0003 2025-04-23 04:00:00+00:00 7.0003 2025-04-23 04:30:00+00:00 28.7898 2025-04-23 05:00:00+00:00 28.7898 2025-04-23 05:30:00+00:00 28.7898 2025-04-23 06:00:00+00:00 28.7898 2025-04-23 06:30:00+00:00 28.7898 2025-04-23 07:00:00+00:0 [... cut ...]
2025-04-22 14:49:32.068208 INFO: Inverter type: SOLIS_SOLAX_MODBUS: inverter module: solis.py
2025-04-22 14:49:32.067788 INFO: Time Zone Offset: 60.0 minutes
2025-04-22 14:49:32.067090 INFO: Local timezone set to GB
2025-04-22 14:49:32.065659 INFO: Debug Categories selected = OTFV
2025-04-22 14:49:32.065299 INFO: Debug Categories selected = OTFV
2025-04-22 14:49:32.064944 INFO: Pre-release version. Enabling debug logging
2025-04-22 14:49:32.064588 INFO:
2025-04-22 14:49:32.064249 INFO: *************** PV Opt Version: v4.0.10-Beta-2 ***************
2025-04-22 14:49:32.063898 INFO:
```

stevebuk1, 2025-04-22:
My system did the same. As you suggest its the Savings Session, the loading of which appears to have mixed UTC and BST, which the code doesn't like. 

I can't immediately see the problem from the logs but will do a beta release with extra debugging to catch it next saving session. 

If your system hasnt recovered just add a carriage return to pv_opt.yaml to get it up and running again.

stevebuk1, 2025-04-22:
v4.0.10-Beta-3 adds additional logging when a savings event is detected. No changes required to debug logging settings (it will log regardless).

swests, 2025-06-04:
Log…

`2025-06-04 16:05:15.853946 WARNING pv_opt: ------------------------------------------------------------
ValueError: Both dates must have the same UTC offset
raise ValueError("Both dates must have the same UTC offset")
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 6875, in slice_locs
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
start_slice, end_slice = self.slice_locs(start, end, step=step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 6662, in slice_indexer
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return Index.slice_indexer(self, start, end, step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/datetimes.py", line 682, in slice_indexer
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
indexer = labels.slice_indexer(slice_obj.start, slice_obj.stop, slice_obj.step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1443, in _get_slice_axis
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self._get_slice_axis(key, axis=axis)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1411, in _getitem_axis
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self._getitem_axis(maybe_callable, axis=axis)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1191, in __getitem__
~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
return getattr(section, self.name)[new_key]
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1089, in _getitem_lowerdim
^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self._getitem_lowerdim(tup)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1368, in _getitem_tuple
^^^^^^^^^^^^^^^^^^^^^^^^
return self._getitem_tuple(key)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1184, in __getitem__
~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
df.loc[event_start:event_end, "unit"] += event_value
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 399, in to_df
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
imp_df = self.tariffs["import"].to_df(start=start.floor("30min"), end=end, **kwargs)
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 673, in net_cost
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cost_today = self.contract.net_cost(grid_flow=grid, log=self.debug, day_ahead=False, sum=False)
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1424, in _cost_actual
self._cost_actual()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 671, in initialize
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^
response = future.result()
File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
await utils.run_in_executor(self, init)
File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
2025-06-04 16:05:15.853855 WARNING pv_opt: Traceback (most recent call last):
2025-06-04 16:05:15.851088 WARNING pv_opt: ------------------------------------------------------------
2025-06-04 16:05:15.851019 WARNING pv_opt: Unexpected error running initialize() for pv_opt
2025-06-04 16:05:15.850925 WARNING pv_opt: ------------------------------------------------------------
2025-06-04 16:03:51.801447 WARNING pv_opt: ------------------------------------------------------------
ValueError: Both dates must have the same UTC offset
raise ValueError("Both dates must have the same UTC offset")
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 6875, in slice_locs
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
start_slice, end_slice = self.slice_locs(start, end, step=step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 6662, in slice_indexer
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return Index.slice_indexer(self, start, end, step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/datetimes.py", line 682, in slice_indexer
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
indexer = labels.slice_indexer(slice_obj.start, slice_obj.stop, slice_obj.step)
File "/usr/lib/python3.11/site-packages/pandas/core/indexing.py", line 1443, in _get_slice_axis
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 16 lines cut ...]
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 399, in to_df
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
imp_df = self.tariffs["import"].to_df(start=start.floor("30min"), end=end, **kwargs)
File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 673, in net_cost
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cost_today = self.contract.net_cost(grid_flow=grid, log=self.debug, day_ahead=False, sum=False)
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1424, in _cost_actual
^^^^^^^^^^^^^^^^^^^
cost_today = self._cost_actual().sum()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2619, in optimise
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
self.optimise()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2312, in optimise_event
^^^^^^^^^^^^^^
return f(*args, **kw)
[... cut ...] object at 0x7f8c15231a10>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2025-06-04T15:03:31.234963+00:00', 'context': {'id': '01JWXQG1N2CES8HKAHJF52WQKG', 'parent_id': None, 'user_id': 'f90d56dbc5da41c6bb0ea75491d9f50e'}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2025-06-04 16:03:51.799992 WARNING pv_opt: Unexpected error in worker for App pv_opt:
2025-06-04 16:03:51.799910 WARNING pv_opt: ------------------------------------------------------------ `

swests, 2025-06-04:
Probably need to tell Octopus that their API needs a small fix :-)

Pyinthesky99, 2025-06-05:
Me too!! I will install v4.0.11-Beta-1

stevebuk1, 2025-06-05:
> Actually, I wonder if the issue is in the Bottlecap Dave add-on. I sat and manually drove the graphQL for the Octopus API and got this…

The BottlecapDave integration is delivering its times in local time, but from your grab the Octopus API works in UTC.  Its not an error as such as such as long as everything stays consistent and that is what is expected. However, one only finds out by error if something is treated as UTC but then an offset is applied when BST kicks in, which is what has happened here. 

I think even with savings events being loaded with BST, the underlying functionality works, in that the savings event price gets correctly loaded into the table of planned 1/2 hour slots (which is usually somewhere between 24 and 48 hours long depending on the time of day). The issue lies within a check that ensures the savings event does lie with the 24/48 hour period. If it doesn't, the savings event start and end is modified so it falls within the available range. This check is made on every calculation - base cost, optimsed cost, etc etc and the instance I caught was one where the end time was 10am but with the savings event being 8pm to 8.30pm. In that case the savings event end time is modified, but its modified to UTC. So we have a saving event start with BST and and savings event end with UTC, which isnt handled due to mixed time offsets. (`ValueError: Both dates must have the same UTC offset`)

04/06, 11:30:54 INFO: event_start = 2025-06-04 20:00:00+01:00
04/06, 11:30:54 INFO: event_end = 2025-06-04 10:00:00+00:00

The fix in Beta-1 is to simply convert the savings event times into UTC first. 

However I'm not convinced its the full fix as its clear from the above the end time is before the start time. We'll need another savings event to test whether that is then a problem.

stevebuk1, 2025-06-30:
Yes please to the logs, if there were issues found would have added extra entries to the log without needing debug switched on. 

Re SolisCloud, I'm meaning to do this on a seperate development Rpi but I've got the opposite problem to you - Solax / Waveshares are stable but my SolisCloud logging is intermittent and I've lost inverter control via the Solis app, so need to fix both of these before giving the SolisCloud functionality a whirl.

Pyinthesky99, 2025-06-30:
Just for a bit of fun/in the interest of research, I flicked into Optimise Discharging at just gone 6 and it discharged until 7. I don't know if the timings were anything to do with the Saver Session or if that would have been the plan anyway. Usually Optimise Discharge tends to set a plan to start dumping the batteries around 4:30 pm.

[pv_opt.log](https://github.com/user-attachments/files/20985208/pv_opt.log)

lazy-pete, 2025-06-30:
Logs from today - not sure how far back they go, but I was fighting with it around 1400 ish I think 

[main.log](https://github.com/user-attachments/files/20986020/main.log)
[error.log](https://github.com/user-attachments/files/20986021/error.log)
[pv_opt.log](https://github.com/user-attachments/files/20986019/pv_opt.log)

swests, 2025-06-30:
Hi @stevebuk1   just checked (been travelling all day) and with 4.0.11-beta1 no errors with the saving session

stevebuk1, 2025-06-30:
Checked my logs and no issues. 

The offer was 104 Octopoints (13p) per kWh for any reduction on normal usage. Pv_opt added 13p to the export price and 13p to the import price which makes sense. 

Apparently I now have the ability to release to production, so will attempt a 4.0.11 release with all the extra debugging removed.

stevebuk1, 2025-06-30:
@fboundy unfortunately the PR failed again, was careful this time to ensure all version numbers had Beta removed, the error message is something about local and remote repositories being different but my locals are the same as remote, perhaps its referring to your local.

stevebuk1, 2025-07-02:
In the absence of being able to release v4.0.11 formal, I've released v4.0.11-Beta-4 which removes all the logging thats generated during a savings session. Will close the issue when v4.0.11 formal is released (via PR).