stevebuk1/pv_opt #403: pv_opt not returning to Idle after Savings Event

Pyinthesky99, 2025-04-22:
**Describe the bug**
After detecting a Savings event at noon 22/4/25 the optimization cycle stopped at 12:00:25     INFO: Optimisation Summary  Base cost.                       

**To Reproduce**
None known

**Expected behavior**
Cycle to return to Idle

**Desktop (please complete the following information):**
 - OS: Win 11
 - Browser Edge
 

**Smartphone (please complete the following information):**
 N/A

**Additional context**
After detecting a Savings event at noon 22/4/25 the optimization cycle stopped at 
12:00:25     INFO: Optimisation Summary
12:00:25     INFO: --------------------
12:00:25     INFO: 
12:00:25     INFO:   Base cost:                                -67.0p

This continued with all cycles from 12:30 until 18:30 (am on 30 min cycle frequency), the Saver Session was due to end at 19:00. The 19:00 cycle, and all cycles since have stopped at getting consumption data.

19:00:00     INFO:     Total consumption from 22-Apr 00:00 UTC to 23-Apr 23:30 UTC:
19:00:00     INFO:     Total consumption: 22.0 kWh

which is earlier in the cycle than the previous stoppages mid cycle.

[error.log](https://github.com/user-attachments/files/19856282/error.log)
[pv_opt.log](https://github.com/user-attachments/files/19856281/pv_opt.log)

Am still running Version: v4.0.9-Beta-10

stevebuk1, 2025-04-22:
My system did the same. Its related to the savings event today, the loading of which appears to have mixed UTC and BST, which the code doesn't like. 

I can't immediately see the problem from the logs but will do a beta release with extra debugging to catch it next saving session. 

Add a line to pv_opt.py and save, which will trigger a restart and get you up and running again.

stevebuk1, 2025-04-22:
v4.0.10-Beta-3 adds additional logging when a savings event is detected. No changes required to debug logging settings (it will log regardless).

Pyinthesky99, 2025-04-23:
> 
> Add a line to pv_opt.py and save, which will trigger a restart and get you up and running again.
>
Assume you meant config.yaml

Will load Beta-3

Thanks for the info.

Pyinthesky99, 2025-04-23:
PS  Only Beta 2 available via HACS

stevebuk1, 2025-04-23:
Config.yaml will also work!

stevebuk1, 2025-04-25:
Closing, as this issue is a duplicate of #402.

closed 2025-04-25 by stevebuk1