stevebuk1/pv_opt #405: Incorrect Timing of Slots

Pyinthesky99, 2025-06-08:
**Describe the bug**
The optimizer, and I, seem to get confused between BST and UTC. 
Slots were scheduled at 6 am to charge at   
08-Jun 07:00 BST - 08-Jun 07:30 BST  Power:  3400W  SOC:   58% ->   74%  
08-Jun 07:30 BST - 08-Jun 08:00 BST  Power:  3000W  SOC:   74% ->   89%  
INFO:   08-Jun 08:00 BST - 08-Jun 08:30 BST  Power:  2700W  SOC:   89% ->  100%

which in itself is odd as that would put 16% into the battery at 1.06p per KWh (7.00 BST) 15% at 0.00p (7.30 BST) and only 11% at -0.02 (8.00 BST)
but then at 06.10 it started to charge at 3.53p!

**To Reproduce**
Steps to reproduce the behavior:
None known

**Expected behavior**
Most charge to be scheduled at the cheapest time

**Screenshots**
I

[error.log](https://github.com/user-attachments/files/20643659/error.log)
[pv_opt.log](https://github.com/user-attachments/files/20643660/pv_opt.log)

**Desktop (please complete the following information):**
 - OS: Win 11
 - Browser Edge
 

**Smartphone (please complete the following information):**
 N/A

**Additional context**
Add any other context about the problem here.

Pyinthesky99, 2025-06-15:
Same again this morning, optimizer scan in log says 7am is 6.6150p

2025-06-15 07:00:00+00:00  1314.80   150.837674       0.5              -1163.962326    2685  -2685.000000  7265.7   8487.4 -2685.054945  1521.0   75.684375   88.410417   6.6150    15.0        0       3
2025-06-15 07:30:00+00:00  1629.10   134.502644       0.5              -1494.597356    2370  -2370.000000  8487.4   9565.8 -2370.109890   876.0   88.410417   99.643750  10.2270    15.0        0       4
 but Octopus times says it is 6.62p at 8am

[image]

[error.log](https://github.com/user-attachments/files/20744658/error.log)
[pv_opt.log](https://github.com/user-attachments/files/20744659/pv_opt.log)

stevebuk1, 2025-06-18:
I don't think theres anything wrong with timezones. 

There is something up with the charge algorithm in that it is only setting the batteries to charge for what it needs from the grid, rather than taking into account of any available solar. 

The three powers that you are getting, namely 3400W, 3000W, 2700W are reducing because the sun is coming up and it expects to get 646W, 982W and 1280W from solar respectively. If you sum these together you get 4kW for all slots, which is what the inverter should be set to for any charging slots allocated (if you are charging in a slot its because its cheap, so 99% of the time you'll want to charge at full rate) unless the price of the slot is exactly the same as any others (not the case in your example). 

The charging at 6.10BST is occurring because it thinks doing it will save you 3p over the course of the day. Its true that if it charges here then the battery is already full in the cheapest slot, so theres definitetly a question whether its a true cost saving. However it might well be, as the sun is well up by that point so it may well be better to export the solar available in the cheapest slot (its 991W from solar = 1280W and house consumption of 289W) and getting 15p for it, rather than doing charging in the cheapest slot and getting 0.02p for it. In any case if we aren't getting full rate charging its a 2nd order effect. 

The question on how solar is added or substracted from charge plans has come up before and although I thought it was fully sorted it appears not. I'm going to struggle to solve this as I don't have an export tariff, so at this time of year Pv_opt is essentially dormant for me.

stevebuk1, 2025-06-18:
> Same again this morning, optimizer scan in log says 7am is 6.6150p

Most of the internal PV_opt logging works in UTC, which doesnt move backwards and forwards. 

The inverter works in local time, so conversions have to be made. 

Octopus - well it varies! 

Anyway, at this time of year, BST is one hour ahead of UTC, for the other six months at GMT, theres no difference (GMT and UTC are numerically the same. 

Re this line:

`2025-06-15 07:00:00+00:00 1314.80 150.837674 0.5 -1163.962326 2685 -2685.000000 7265.7 8487.4 -2685.054945 1521.0 75.684375 88.410417 6.6150`

2025-06-15 07:00:00+00:00 is in UTC which equates to 8.00 BST, which does have a price of 6.62 in your colour snapshot (the 6.615 in the above line, rounded up).

stevebuk1, 2025-06-18:
I forgot I added some debug logic to the equation that does the forced charge calculations. 

Set the debug_categories line in config.yaml to have a C in it, as follows:

`debug_categories: OC `

Please post a log when you get a charge slot anytime from 7am onwards.

stevebuk1, 2025-06-18:
For my own records the issue around solar summing was #259

Pyinthesky99, 2025-06-19:
Thanks for the exhaustive explanation, as ever.
I have changed debug_categories: from OS to OC, hope that's OK, am not holding my breath for a charge slot!

Pyinthesky99, 2025-06-19:
Having said that, I am still not convinced it didn't start charging at 8.00am BST. (7.00am UTC)
This is a snip of the HA History of SOC for that day, 15th June.

[image]

It started charging at 7.00am so we assume the HA History is UTC? Home Assistant console is UTC, an hour behind us on good old BST (GMT[UTC] + 1hr.

But, it's just gone 2pm here (BST) and the History is saying..

[image]

So that would suggest History is not UTC but BST.

I think I will have to set my alarm clock for the next charge time, thankfully it just tells the time by the big hand and the little hand and not what zone I am in :-)

stevebuk1, 2025-06-19:
> Having said that, I am still not convinced it didn't start charging at 8.00am BST. (7.00am UTC) This is a snip of the HA History of SOC for that day, 15th June.
> 
> [image]
> 

Ok so finding the line that you quoted in the log shows 3 charging slots (as indicated by the large non-zero numbers (in W) in the 7th column:
7:00-7.30BST
8:00-8:30BST
8:30-9:00BST

```
2025-06-15 06:00:00+00:00   598.75   186.983036       0.5               -411.766964    3401  -3401.000000  5358.3   6905.8 -3401.098901  2989.0   55.815625   71.935417  10.8675    15.0        0       1
2025-06-15 06:30:00+00:00   992.00   200.994571       0.5               -791.005429       0   -791.005429  6905.8   7265.7  -790.989011    -0.0   71.935417   75.684375  12.3480    15.0        0       2
2025-06-15 07:00:00+00:00  1314.80   150.837674       0.5              -1163.962326    2685  -2685.000000  7265.7   8487.4 -2685.054945  1521.0   75.684375   88.410417   6.6150    15.0        0       3
2025-06-15 07:30:00+00:00  1629.10   134.502644       0.5              -1494.597356    2370  -2370.000000  8487.4   9565.8 -2370.109890   876.0   88.410417   99.643750  10.2270    15.0        0       4
```

Checking the final outcome also shows the same:

00:30:21     INFO: Optimal forced charge/discharge slots:
00:30:21     INFO:   15-Jun 07:00 BST - 15-Jun 07:30 BST  Power:  3400W  SOC:   56% ->   72%  
00:30:21     INFO:   15-Jun 08:00 BST - 15-Jun 08:30 BST  Power:  2700W  SOC:   76% ->   88%  
00:30:21     INFO:   15-Jun 08:30 BST - 15-Jun 09:00 BST  Power:  2400W  SOC:   88% ->  100%  

> It started charging at 7.00am so we assume the HA History is UTC? 

No, HA history is in local time. Your plot lines up with the charge plan above. You can see where the charging slows down between 7:30 and 8:00 (BST) because its now only charging from solar, then speeds up again at 8am (BST)

> 
> But, it's just gone 2pm here (BST) and the History is saying..
> 
> [image]
> 
> So that would suggest History is not UTC but BST.

Correct! 
> 
> I think I will have to set my alarm clock for the next charge time, thankfully it just tells the time by the big hand and the little hand and not what zone I am in :-)
> 
> Edit.. In their wisdom, Octopus have just sent me a statement for the 1st to 8th July, despite numerous requests to just send them out on the 1st of each month, fortunately that gave me a chance to look up the numbers on my opening post and the times and rates correlate...
> 
> [image]

Pyinthesky99, 2025-06-20:
> Having said that, I am still not convinced it didn't start charging at 8.00am BST. (7.00am UTC) This is a snip of the HA History of SOC for that day, 15th June.
> 
Indeed, it didn't start charging at 8.00BST, it started at 7.00BST because that was the time it had calculated to start!!

Thanks again for the info, I should trust the system :-)

There is a slot scheduled for tomorrow morning but I think it may not be relevant, will keep you posted.

Pyinthesky99, 2025-06-22:
Saturday morning's slots got cancelled but this morning we may have captured what we are after.

[pv_opt.log](https://github.com/user-attachments/files/20851874/pv_opt.log)

[pv_opt.log1.txt](https://github.com/user-attachments/files/20851887/pv_opt.log1.txt)

stevebuk1, 2025-06-22:
Negative prices all morning made for a good logging run for the issue, thanks. 

(due to poor memory I forgot to plug my EV in last night so set a charge target time of 11.00, usually its hit and miss whether I get any cheap slots on IOG if I do this but a quick check of Octopus Watch showed me why it was no issue!)

stevebuk1, 2025-06-23:
So the algorithm was deliberately changed back in June last year to substract solar power from the maximum charge rate:

[image]

Whilst I think this makes sense in High Cost Swaps I'm less convinced in Low Cost Charging, for the reasons I gave earlier. However, whichever is optimum should result in the best cost saving (the whole point of Pv_opt) so the answer is to run the two in parallel for a while and see what one offers the best savings. 

The optimiser already runs four times to generate base, optimised charging, optimised pv export and optimised discharge so I've added a 5th called "Optimised PV Export (alt)", which is a version of Optimised Pv Export that doesnt substract solar, as follows:

[image]

v4.0.11-Beta-2 released in HACS to do the trial. 

In config.yaml, modify debug categories to "AO", the A is the logging of all 5 plans separately, this will allow us to compare the cost savings with subtract solar on and substract solar off. 

Note that at the moment `Optimised PV Export (alt)` is for logging only and is not selectable as an option, i.e. inverter setup is unchanged.

Pyinthesky99, 2025-06-24:
Thanks for the info and change. I have loaded and it seems to be running ok.

[image]

I did have a burst of charge at 9:30 this morning, unfortunately before the download, will keep you posted.

Pyinthesky99, 2025-06-24:
Apologies for being a PIA. I inadvertently clicked on Pass Threshold and it went from 4 to .5! Actually, it was advertently as the saving was 5p so wondered what it would do if I took it up to 5 or 6! I have tried to adjust the slider back to 4 but it wont go past 1. pv_opt.py states default is 4, config.yaml states overwrite_ha_on_restart: true so have restarted by putting a CR in config.yaml, then restarted AppDaemon and then restarted HA but Pass Threshold is stubbornly stuck on 1.

stevebuk1, 2025-06-24:
Getting the limits of an entity to update once MQTT originally generates it is impossible. 

There are two methods, published here:

https://github.com/fboundy/pv_opt/releases/tag/v4.0.9-Beta-10

Pyinthesky99, 2025-06-27:
These are some snips taken around 7pm Friday with changes in Consumption. the (alt) profile marginally better with higher usage.

[image]

[image]

[image]

My daily consumption is probably on the lower side.

[pv_opt-1.log](https://github.com/user-attachments/files/20953472/pv_opt-1.log)

[pv_opt.log](https://github.com/user-attachments/files/20953455/pv_opt.log)

Will upload log again after the event

Hope this helps

Pyinthesky99, 2025-06-28:
Here are this mornings logs, they tend to get rather full, rather quickly!

[pv_opt.log](https://github.com/user-attachments/files/20959997/pv_opt.log)

[pv_opt1.log](https://github.com/user-attachments/files/20959999/pv_opt1.log)

stevebuk1, 2025-06-29:
Beta-2 fixed the charge rates and when the timings are right, full rate charging is beneficial over subtracting solar. 

The bigger problem remains, that whilst the cheapest slot is processed first it doesnt seem to get added before then calculating the costs of the more expensive slots, which means if slots get progressively cheaper the battery is already fully charged. 

I've released v4.0-11-Beta-3 with further logging on this, can you change debug_categories to AOC, i.e. `debug_categories: OAC`

Pyinthesky99, 2025-06-30:
Loaded, that's a lot of logging, don't mind a drop of AOC though.

Pyinthesky99, 2025-07-02:
[image]

Would have thought it would have topped up batteries as solar pretty poor today. Had to set Optimise Discharging (not normally set) and set use Consumption History (I usually set daily usage a trifle high to err on the side of caution) before it added it to the plan. Plan does seem odd as it is least Watts at the cheapest rate.

[image]

[pv_opt.log](https://github.com/user-attachments/files/21017357/pv_opt.log)

stevebuk1, 2025-07-02:
> Would have thought it would have topped up batteries as solar pretty poor today.

Pv_opt is only going to charge if the batteries run out at any point, and even then it will only do it if the electricity rate at the point that the batteries run out is higher than elsewhere. Neither of those things happened. 

> Had to set Optimise Discharging (not normally set) and set use Consumption History (I usually set daily usage a trifle high to err on the side of caution) before it added it to the plan. 

I'd toyed with adding a numeric target % that the batteries will not fall below so act as a battery reserve, there in case of unscheduled  washing machine runs etc. (Predbat calls this "best_soc_keep". However, you can achieve exactly the same thing by setting a consumption margin. 30% is the figure I use. 

> Plan does seem odd as it is least Watts at the cheapest rate.

As per the title of this issue, Pv_opt subtracting solar from charge rates and the cheapest slot happens to have the most solar, so the least wattage. Today however, doing full rate charging in the cheapest slot did meet the 1p theshold for considering the slot, but by the time all the slots were considered the saving was only 3.2p, so no slots were scheduled. Whereas doing reduced rate charging in the cheapest slot didnt meet the 1p threshold in that slot, but charging in the more expensive slots did reduce the overall cost (as it was more than 4p), as your plan summary shows. So for today, the right answer was to do the reduced rate charging. I think I can predict where this is going to end up - we've probably reached the limits of the algorithm to cope with all scenarios of consumption and solar and the differences are a few pence at most.

stevebuk1, 2025-07-02:
I forgot to answer my own question:

> The bigger problem remains, that whilst the cheapest slot is processed first it doesnt seem to get added before then calculating the costs of the more expensive slots, which means if slots get progressively cheaper the battery is already fully charged by the time the cheap slots are reached.

The logging we've now got shows that the cheapest slot is indeed added first (as long as it saves >1p) before considering the more expensive slots, so its all fine. 

I suggest logging a few more runs at cheap slots but I have a nasty feeling all of this is going to end up as 50/50 which version of Optimised Pv Export will save the most money depending on pricing and solar on any one particular day.

Pyinthesky99, 2025-07-03:
Don't know if you would like me to let you know whenever I notice a difference in (alt)

[image]

[pv_opt.log](https://github.com/user-attachments/files/21041092/pv_opt.log)

stevebuk1, 2025-07-04:
> Thought I had as you had mentioned it before, but it had reset back to 10. Don't think it can be set it config.yaml

Add   `consumption_margin: 30` to config.yaml to permanently configure to 30%.

stevebuk1, 2025-07-04:
> > Pv_opt is only going to charge if the batteries run out at any point, and even then it will only do it if the electricity rate at the point that the batteries run out is higher than elsewhere.
> 
> Exactly what I was trying to avoid.

But avoiding that costs you money. Pv_opt may well plan to allow the batteries to run out if they are due to run out at the cheapest part of the day, and then solar cuts in to charge them before it gets expensive again.  The only point of charging batteries from the grid is to utilise whats in them them at a more expensive time. At this time of year on Agile, it may well be lowest cost to leave them flat for periods of time.

Pyinthesky99, 2025-07-05:
[image]

[pv_opt.log](https://github.com/user-attachments/files/21074200/pv_opt.log)

Pyinthesky99, 2025-07-06:
[image]

[pv_opt.log](https://github.com/user-attachments/files/21088077/pv_opt.log)

stevebuk1, 2025-07-07:
Closing issue, as its an algorithm limitation as opposed to a bugfix.

closed 2025-07-07 by stevebuk1

Pyinthesky99, 2025-07-08:
Thanks for the explanation, it would be simpler if we did without the panels, but not better if we did without the sunshine!

Will look forward to the AddOn.