stevebuk1/pv_opt #410: Issue with IOG car charing slots, causes optimiser to fail

swests, 2025-07-15:
HA : 2025.7.2
Octopus:  16.0.0
PV_opt: 5.0.0 beta 1
.
.
PV_opt Log:

```
2025-07-15 09:26:30.097901 INFO: Failed to load contract in 5 attempts. FATAL ERROR
--
2025-07-15 09:26:18.096889 INFO: Failed to load contact - Attempt 5 of 5. Waiting 2 minutes to re-try
2025-07-15 09:26:18.096183 WARNING: Failed to find tariff from Octopus Energy Integration
2025-07-15 09:26:18.095624 ERROR: 1546: 'charge_in_kwh'
2025-07-15 09:26:18.094681 INFO: Octopus Intelligent Go Smart Charging Schedule is....
2025-07-15 09:26:18.094106 INFO:
2025-07-15 09:26:18.091208 INFO: Current Dispatch Status (On/off) = : off
2025-07-15 09:26:18.089940 INFO:
2025-07-15 09:26:18.089437 INFO: -----------------------------------------------------------
2025-07-15 09:26:18.088886 INFO: Reloading IOG Car slots from Intelligent Dispatching Sensor
2025-07-15 09:26:18.088375 INFO:
2025-07-15 09:26:18.087828 INFO: Charge to Add Value = : 80
2025-07-15 09:26:18.086874 INFO: Found Charge to Add entity: number.octopus_energy_**********_intelligent_charge_target
2025-07-15 09:26:18.084132 INFO: Trying to find Car % Charge to add from Octopus Energy Integration
2025-07-15 09:26:18.083631 INFO:
2025-07-15 09:26:18.083109 INFO: Found Dispatching Sensor: binary_sensor.octopus_energy_**********_intelligent_dispatching
2025-07-15 09:26:18.080134 INFO: Trying to find Octopus Intelligent Dispatching Sensor from Octopus Energy Integration
2025-07-15 09:26:18.079640 INFO:
2025-07-15 09:26:18.079117 INFO: -------------------------
2025-07-15 09:26:18.078601 INFO: Getting Car Charging Plan
2025-07-15 09:26:18.078046 INFO:
2025-07-15 09:26:17.998151 INFO: Reading next day IOG prices from event.octopus_energy_electricity_************************_next_day_rates
2025-07-15 09:26:17.995920 INFO: Reading current day IOG prices from event.octopus_energy_electricity_************************_current_day_rates
2025-07-15 09:26:17.993170 INFO:
2025-07-15 09:26:17.992631 INFO: Downloading IOG pricing information from Octopus Energy Integration
2025-07-15 09:26:17.992101 INFO:
2025-07-15 09:26:17.991555 INFO: Octopus Intelligent Import Entity found: ['event.octopus_energy_electricity_************************_current_day_rates']
2025-07-15 09:26:17.988745 INFO: Trying to find Octopus Intelligent Entities from Octopus Energy Integration:
2025-07-15 09:26:17.836256 INFO: Found export entity event.octopus_energy_electricity_************************_export_current_day_rates: Tariff code: E-1R-OUTGOING-FIX-12M-19-05-13-H
2025-07-15 09:26:17.835279 INFO: Found import entity event.octopus_energy_electricity_************************_current_day_rates: Tariff code: E-1R-INTELLI-VAR-24-10-29-H
2025-07-15 09:26:17.832286 INFO: Trying to auto detect Octopus tariffs:
2025-07-15 09:26:05.831492 INFO: Failed to load contact - Attempt 4 of 5. Waiting 2 minutes to re-try
2025-07-15 09:26:05.830746 WARNING: Failed to find tariff from Octopus Energy Integration
2025-07-15 09:26:05.830200 ERROR: 1546: 'charge_in_kwh'
2025-07-15 09:26:05.829278 INFO: Octopus Intelligent Go Smart Charging Schedule is....
2025-07-15 09:26:05.828722 INFO:
2025-07-15 09:26:05.825786 INFO: Current Dispatch Status (On/off) = : off
2025-07-15 09:26:05.824518 INFO:
2025-07-15 09:26:05.823961 INFO: -----------------------------------------------------------
2025-07-15 09:26:05.823454 INFO: Reloading IOG Car slots from Intelligent Dispatching Sensor
2025-07-15 09:26:05.822929 INFO:
2025-07-15 09:26:05.822386 INFO: Charge to Add Value = : 80
2025-07-15 09:26:05.821394 INFO: Found Charge to Add entity: number.octopus_energy_**********_intelligent_charge_target
2025-07-15 09:26:05.818692 INFO: Trying to find Car % Charge to add from Octopus Energy Integration
2025-07-15 09:26:05.818191 INFO:
2025-07-15 09:26:05.817664 INFO: Found Dispatching Sensor: binary_sensor.octopus_energy_**********_intelligent_dispatching
2025-07-15 09:26:05.814674 INFO: Trying to find Octopus Intelligent Dispatching Sensor from Octopus Energy Integration
2025-07-15 09:26:05.814190 INFO:
2025-07-15 09:26:05.813665 INFO: -------------------------
2025-07-15 09:26:05.813143 INFO: Getting Car Charging Plan
2025-07-15 09:26:05.812602 INFO:
2025-07-15 09:26:04.952833 INFO: Reading next day IOG prices from event.octopus_energy_electricity_************************_next_day_rates
2025-07-15 09:26:04.950664 INFO: Reading current day IOG prices from event.octopus_energy_electricity_************************_current_day_rates
2025-07-15 09:26:04.947750 INFO:
2025-07-15 09:26:04.947222 INFO: Downloading IOG pricing information from Octopus Energy Integration
2025-07-15 09:26:04.946657 INFO:
2025-07-15 09:26:04.946100 INFO: Octopus Intelligent Import Entity found: ['event.octopus_energy_electricity_************************_current_day_rates']
2025-07-15 09:26:04.943300 INFO: Trying to find Octopus Intelligent Entities from Octopus Energy Integration:
2025-07-15 09:26:04.784878 INFO: Found export entity event.octopus_energy_electricity_************************_export_current_day_rates: Tariff code: E-1R-OUTGOING-FIX-12M-19-05-13-H
2025-07-15 09:26:04.783709 INFO: Found import entity event.octopus_energy_electricity_************************_current_day_rates: Tariff code: E-1R-INTELLI-VAR-24-10-29-H
2025-07-15 09:26:04.780483 INFO: Trying to auto detect Octopus tariffs:
[... 533 lines cut ...]
car_slots = self._get_io_car_slots() # Load the car charging slots
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2670, in optimise
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
self.optimise()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2323, in optimise_time
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
Traceback (most recent call last):
 
The above exception was the direct cause of the following exception:
 
KeyError: 'charge_in_kwh'
File "pandas/_libs/hashtable_class_helper.pxi", line 7096, in pandas._libs.hashtable.PyObjectHashTable.get_item
File "pandas/_libs/hashtable_class_helper.pxi", line 7088, in pandas._libs.hashtable.PyObjectHashTable.get_item
File "pandas/_libs/index.pyx", line 196, in pandas._libs.index.IndexEngine.get_loc
File "pandas/_libs/index.pyx", line 167, in pandas._libs.index.IndexEngine.get_loc
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
return self._engine.get_loc(casted_key)
File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 3812, in get_loc
```

swests, 2025-07-15:
Rolled back Octopus to previous version (15.3.0), no change.

stevebuk1, 2025-07-15:
I've also noted some strange behaviours from the Octopus energy integration over the last few days - it permanently reported a charging schedule via planned dispatches despite the car not being plugged in and the Octopus app stating no schedule, which I think is what you are seeing too. 

I didnt charge last night but will need to tonight so will also add debug categories, main aim is to prevent Pv_opt crashes if data is erroneous.

stevebuk1, 2025-07-15:
Also add "T" to the list of debug categories as well as "E", as the error might be in tariff load.

stevebuk1, 2025-07-16:
Theres definitely some issues with V16.0.0 Octopus Energy Integration. 
BottlecapDave has said OE are going to retire the current API endpoint so hes moved over to the replacement in advance:
https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy/issues/1369

And the Predbat guys are reporting bugs specifically to do with charge_in_kwh in v16.0.0:
https://github.com/springfall2008/batpred/issues/2560

It also mention also mentions `planned_dispatches` is going to be deprecated on 27th August, confirmed here:
https://developer.octopus.energy/graphql/reference/queries#api-queries-planneddispatches

I'll look to move over to the replacement `flexPlannedDispatches` but in the meantime suggest rollback to v15.3.0 in the meanime.

stevebuk1, 2026-04-05:
bump. Looks fine but yet to be included in production.