stevebuk1/pv_opt #411: Solis Cloud entities break PV_Opt - where to edit

lazy-pete, 2025-07-22:
**Describe the bug**
A clear and concise description of what the bug is.

I think an update to Solis Cloud Inverter Control has changed some required entities. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Having spent the evening trying to configure HA to work with a Zappi and a new EV I have finally got it running after moving to Beta 5.0.2. However it errors when writing to the inverter. The sought for entities do not seem to exist, as I can only find charge_discharge. I’ve tried editing the yaml to make each id seek the same entity but to no avail. 

**Expected behavior**
A clear and concise description of what you expected to happen.

Succesfully write to inverter. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

[main.log](https://github.com/user-attachments/files/21355761/main.log)
[error.log](https://github.com/user-attachments/files/21355762/error.log)
[pv_opt.log](https://github.com/user-attachments/files/21355763/pv_opt.log)

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

lazy-pete, 2025-07-23:
Thanks Steve.

Whilst poking around for a solution I found a post on the solid cloud integration that eventually led me to some updated files that have yet to make their way into a version update. Manually copying / pasting the custom components into the solis custom components files and then rechecking the control api function resulted in a host of entities appearing, including it seems the various ones i was missing, after which PV_OPT seems to be up and running again. I can't pretend to understand what those files actually changed, but the entities I needed were enabled afterwards.

https://github.com/hultenvp/solis-sensor/pull/483/files#diff-52f2298e5853c98c839950bbb469e6a5d2066c7b9fa98b24d2c6621046296c48

I will look into your suggestion tomorrow night for my own understanding. 

Thanks again for chiming in. 

Pete

stevebuk1, 2025-07-23:
> I’m not sure how to test the actual API for the control integration - I’m guessing g if pv_opt is doing its part it writes to that and thinks the job is done ?

For SolisCloud, Pv_opt simply writes to/reads from the entities that are associated with the SolisSensor integration (https://github.com/hultenvp/solis-sensor). There was a version of Pv_opt that did the writes directly to the SolisCloud API, but Francis then added SolisControl functionality directly into SolisSensor and removed it from Pv_opt. 

In other words, just directly changing the various entities in HA should control the inverter, I'd definitely do that as a first step. 

Pv_opt only uses slot 1 regardless of firmware, so create a dashboard with the following entities:

time.solis_timed_charge_start_1
time.solis_timed_charge_end_1 
button.solis_update_timed_charge_1
number.solis_timed_charge_soc_1                                  
number.solis_timed_charge_current_1         

Then:

1) Set the start and end time entities
2) Press the update timed charge button
3) Set the charge current
4) Set the SOC 

(Its my understanding that setting the SOC is what is needed to activate the slot).

If this doesnt work its a case of raising an issue at https://github.com/fboundy/solis-sensor/issues as stated in https://github.com/hultenvp/solis-sensor. 

PV_opt values of 535% are coming directly from reads of number.solis_timed_charge_soc_1 above. Version 42 of the firmware is 3 slot (6 slot was introduced at 4A). I think thats your problem but SolisSensor should autodetect. Does your inverter front screen allow you to set 6 slots?

stevebuk1, 2025-07-23:
Took a look at your link above and found this:

[image]

I'd agree 42 is arbitary (and wrong). This explains your problem, your inverter is 3 slot but the use of 42 is creating entities for 6 slot.

stevebuk1, 2025-07-23:
I've added this:

[image]

to https://github.com/hultenvp/solis-sensor/pull/483

stevebuk1, 2025-07-24:
> So ultimately, once I'd established the constituent parts worked, it came down to pv_opt not triggering that button press, I think.

The pv_opt logging is pretty clear on button presses (but not on updating time entities I note). Sounds like a mapping error to the button press entity in config.yaml but a Pv_opt log should tell us. A lack of a button press when one is needed should be pretty easy to track down. Please post config.yaml alongside pv_opt.log so I can check the mapping.

stevebuk1, 2025-07-25:
Thanks for the logs and the explanation. The log doesn't show an issue with Pv_opt writing to any of the inverter entities but I'd agree the inverter is not being updated. 

The PV_opt README only uses solis-sensor as the integration, and this is capable of both reading the inverter and writing to it, meaning there is no need for https://github.com/mkuthan/solis-cloud-control. (Also Francis wrote the inverter control code for solis-sensor). My cross post was to solis-sensor, so if you arent using it that won't be the problem. I'd still agree its the way to go (using seperate integrations for monitoring and control) but Pv_opt was tested with solis-sensor only, which might be a problem. 

Have you definitely got the inverter control side of solis-sensor switched off? ->

[image]

Finally, with respect to PV_opt, I don't recognise this line in config.yaml `use_soc_based_control: false`. Is this a setting for something else that should be elsewhere? 

I've temporarily plugged my Solis datastick directly into my inverter instead of its usual location on a Waveshare, but its not made a difference in that updates are 15 to 30 minutes apart rather than the expected 5 and I have no inverter control via the Solis app (this all used to work fine). I've got a new stick on order so when that arrives I'll see if that solves things and then I'll install https://github.com/mkuthan/solis-cloud-control on my development Pi to get this solved.

lazy-pete, 2025-07-25:
Thanks - I have to use both integrations I think, because the entities provided by control aren’t the ones listed in the pv_opt config yaml - the write slots are start / end combined as opposed to separate start and end entities. Everything needed is in the sensor integration, apart from the fact that the mode switch always reports an unknown state , thus I have to use the mode entity in the control integration (pv_opt errored from the other one being ‘unknown’).

The PV_opt Readme references BOTH the Solis-sensor AND the Solis-control integrations so I presumed using elements of each was ok - it did seem to get me to the point of generating a charge plan reliably.

That extra line of code in the yaml is something chatGPT suggested to dispense with the looking for a Soc entity state when one didn’t exist (because I’m on 3 not 6 slot firmware). May well be redundant now.

I was mostly puzzled as to why PV_OPT created a charge plan, counted down how close it was to needing to apply it, with feasible looking times and charge currents, but then just skipped by and put currents back to 0. There seemed to be some times near the initial planned start where pv_opt thought the inverter was already charging, which it wasn’t. My car may have been charging on the Zappi, but I think that is detectable as a separate entity from the inverter so all in all confused.

I’ve put in into read only meantime and will try the Solis control / sensor (with controls enabled) directly to check at least that works. As I’m on IOG now at least charge slots are predictable, though it would be nice to leverage the intelligent charge slots as well. I think cyclic charge / discharge is against their T&C’s anyway.

If you manage to get it working / need any more logs or testing please let me know. My WiFi connection to the Solis-cloud via the Solis supplied stick seems much more stable than the Modbus Waveshare arrangement I was using.

Cheers for putting time into this.

P

image.png (view on web)<https://github.com/user-attachments/assets/92019bfe-1461-479d-9c20-eed06d8534cb>

stevebuk1, 2025-07-26:
> Maybe I just need to get under the floor and run Ethernet under the garden to the garage.

My inverter is in the garage also, with no ethernet connection to the house. Just an idea but I use a Powerline adaptor to connect a Waveshare in the garage to the RPi running Pv_opt in my house, it also gives a Wifi hotspot so can connect the Solis datastick as well.  Ignore the nonsense about Mesh Wifi being a new thing - all Wifi devices are more than capable from moving from access point to access point based on whats the strongest, so simply set your new Powerline adaptor to the same SSID as everything else and you'll have seemless wifi in the garage as well as everywhere else. I use this one:

https://www.amazon.co.uk/dp/B08QSK84JL?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_1&th=1

stevebuk1, 2025-07-26:
> I won’t hear back from them until Monday when the Puredrive support opens. Maybe multiple things going on at once.

It was my gut reaction that you've got two things intefering with each other. I remember having the same issue when I was running predbat in parallel and it had come out of readonly mode without me noticing!

stevebuk1, 2025-07-26:
> I don’t quite have enough tech awareness to set up the systems needed to allow the Waveshare to work with two masters (ie my Solis WiFi stick and the Modbus elfin thing).

Yes beyond me too (and I used to work with RS485 professionally as an interface medium in an aircraft). The waveshare route saves all this as there is 1 RS485 link per Waveshare and they all connect together over ethernet.

stevebuk1, 2025-07-26:
Forgot to add the image from the setup I use

[image]

lazy-pete, 2025-07-30:
Now even more confused. 

Puredrive (Now Duracell I think) have removed any tariff info from my account so should no longer be calling the Solis API. 

I can control the inverter remotely using the Solis control integration but it seemed to be overwritten a bit later. I think this may be a function of the button press updating from both control and sensor integrations (I have an automation running to ensure the button is pressed 5s after a change in the charge time elements, as that seemed to be a previous fail point. As a result I’ve disabled the charge time entities of the control integration and that now seems to work. 

However, the main log has an error on the state options of the inverter Energy Storage Control Switch I’ve mapped in config.yaml (because the default only exists in the sensor integration and always has a state of unknown. 

PV_Opt runs none the less and tells
Me it’s writing times to the inverter, and that it’s charging, however it is not. 

What I think is happening is it’s somehow reading the Octopus charge slots for my car (via Zappi) and writing those to the inverter, with a current, but that doesn’t seem to be actually causing it to charge. 

Logs and screenshots below. 

I’ve also put in a pic of the Modbus setup I was using, which I may have to try again. Perhaps adding the car has been one thing too many, but it would be nice to grab those extra charge slots. 

Any ideas ?

[main.log](https://github.com/user-attachments/files/21518821/main.log)
[error.log](https://github.com/user-attachments/files/21518822/error.log)
[pv_opt.log](https://github.com/user-attachments/files/21518820/pv_opt.log)

[image]
[image]
[image]

lazy-pete, 2025-07-31:
Code inserted and Appdaemon restarted at 1749 ish.

Logs below. 

[error.log](https://github.com/user-attachments/files/21535015/error.log)
[main.log](https://github.com/user-attachments/files/21535016/main.log)
[pv_opt.log](https://github.com/user-attachments/files/21535017/pv_opt.log)

lazy-pete, 2025-07-31:
Solis Control integration first 

[image]
[image]
[image]

lazy-pete, 2025-07-31:
Now Solis sensor (with control option active)

This one is closer to the correct list, but always reports a status of unknown which stops PV-opt running. 

There is a comment in one of the Hultenvp integrations regarding the text vs numerical reporting. 

https://github.com/hultenvp/solis-sensor/issues/443

[image]
[image]
[image]
[image]
[image]
[image]

lazy-pete, 2025-08-01:
Thanks. 

Bit odd overnight. It did charge the battery between about 0100 and 0330, but then allowed it to discharge until the solar kicked in and charged it back up. I haven’t worked out all the algorithm stuff but sort of assumed it should hold 100% until end of off peak tarrif time, and then discharge until enough PV to pick up the load, then use the rest in the evening before off peak kicked in again, but PV-opt may well be smarter than me on that. 

Currently it says it’s charging, is in a time slot that should be charging, but isn’t actually charging. 

Bit confused. 

At least it’s doing something though !

Glad your new stick works. Will be interested to see if you can make it work with only the entities Solar-Control gives you. As I said, I’ve needed to use both to access the combination of entities that config.yaml asks for. 

Am going to turn of my button press in status change for tonight and see if it still works. 

Cheers,

Pete 

[image]
[image]
[image]
[image]
[image]

[error.log](https://github.com/user-attachments/files/21554077/error.log)
[main.log](https://github.com/user-attachments/files/21554075/main.log)
[pv_opt.log](https://github.com/user-attachments/files/21554076/pv_opt.log)

stevebuk1, 2025-08-01:
> Bit odd overnight. It did charge the battery between about 0100 and 0330, but then allowed it to discharge until the solar kicked in and charged it back up.

By looking at the battery SOC on each 10 minute run, I'd concur it did exactly that. 

> I haven’t worked out all the algorithm stuff but sort of assumed it should hold 100% until end of off peak tarrif time, and then discharge until enough PV to pick up the load, then use the rest in the evening before off peak kicked in again, but PV-opt may well be smarter than me on that.

The plan last night was indeed to charge to 100% in the cheap rate, then hold it to 5.30. It just didnt do that. It did actually tell the inverter to do that, but it went wrong at 3.30am and didn't respect it. 

I think there is something going on with your setup re GMT and BST. The charge slots are working when they are hours long (like the 1.00am to 3.30pm one was) but when it commands a half hour slot the inverter is not charging. The reason I think this is might be the problem is because your SolisCloud app screenshot above states the last data reporting time was 20:06:30 (UTC+00:00) when your phone time is 21:06 i.e a full hour ahead. My SolisCloud phone app right now reads as `21:48 (UTC+00:00)` and its 21:53 as I read this. What is the time on your inverter front screen actually saying? 

In the app, there are two places you need to set the timezone:

1) "Me" Icon bottom right, settings, time zone, set to (UTC+00:00) Europe/Dublin and ensure the Daylight saving time slider is set
2) Overview Page, Modify Plant Info, Time Zone, set to UTC+00/Europe/Dublin, and ensure the Daylight saving time slider is set.

lazy-pete, 2025-08-01:
Thanks again Steve. 

You are correct that my inverter times did not have BST selected - it was my understanding from the past that it was preferable to leave the inverter as UTC and PV-opt compensated for that when righting times to the inverter. I am perhaps mistaken. I have enabled BST now in both locations. Do I need to manually turn that on / off each time the clocks change ? I don’t think Solis did that automatically in the past, hence I thought an external workaround had been applied. 

Screenshots below from the Solis-cloud-control integration - you will see I have slot 1 disabled as the config.yaml wanted separate start / end entities which were only available in Solis-sensor, as you have noted. Screenshot of slot 2 so you can see how it is configured. Then below are screenshots from Solis-sensor (which also allows control entities once the appropriate flag is selected on setup). Again, you have spotted the difference in config of the energy control switch between the two, hence my config.yaml pointing at some entities of both integrations. This is how I am writing to the correct time slots - I have no Modbus connection at present - the integrations are disabled / deleted, although the WiFi dongle for Solis cloud connectivity is routed through the same box my Modbus WiFi routes. I could plug the WiFi dongle directly into the inverter, but can’t see where any Modbus commands would be coming from anyway. I never got the HA Core Modbus working anyway and don’t have PV-opt configured to use it. I had intended to setup some software to allow my setup to work properly as a ‘two masters / one slave’ so I could have Solis cloud monitoring and Modbus control, but it was too complex for my technical (in)ability. Running both together resulted in conflicts so for a long time I used only Solax Modbus, which was great and periodically failed. I think it was too sensitive to network instability and I was forever having issues with it - gibberish energy monitoring values / stopping working randomly and dropping out. The Cloud control ‘feels’ more stable, albeit it doesn’t quite work !

I’ll post some more logs after the DST changes (and disabling my time update monitored button press automation) in the morning. 

Some sort of merging of the functionality of Solis-control and Solis-sensor seems a good idea, which I thought is what adding the control option to sensor was supposed to allow, which would work if it wasn’t for the status issue of the energy control switch I think (for my 3 slot model, since the hmi flag corrected that). I note the most recent update to Solis-sensor 3.13.0 (which I have not applied while working through this) is related to the HMI flag. 

[image]

Cloud-control

[image]
[image]
[image]
[image]

Solis-sensor

[image]
[image]
[image]
[image]
[image]
[image]
[image]

stevebuk1, 2025-08-01:
> You are correct that my inverter times did not have BST selected - it was my understanding from the past that it was preferable to leave the inverter as UTC and PV-opt compensated for that when righting times to the inverter. I am perhaps mistaken.

I am certain Pv_opt loads the inverter with local time values, so the clock on the inverter needs to be local time. There were definitely bugs with inverter writes in the past but now its a case that Pv_opt calculates in UTC but writes to the inverter in local time. 

> Do I need to manually turn that on / off each time the clocks change ? 

No, you shouldn't, mine worked this year with no adjustments but it took a year to get there! 

> I don’t think Solis did that automatically in the past, hence I thought an external workaround had been applied.

I think you are right. Its also only recently the inverter time is now syncd via the app, certainly the Solaxmodbus integration had a button for updating the inverter time which I had to press every few months due to time drift, but now never do.

> Some sort of merging of the functionality of Solis-control and Solis-sensor seems a good idea, which I thought is what adding the control option to sensor was supposed to allow, which would work if it wasn’t for the status issue of the energy control switch I think

Adding the control option to Solis Sensor was intended to remove the need for Solis-Cloud-Control completely, which it would have done save the issue with the energy storage switch. Unfortunately the owner of Solis Sensor has asked issues to be raised directly on Francis's fork, and I don't see these being addressed.

lazy-pete, 2025-08-01:
Yep - also saw little action on that, hence chancing it by raising my
problems on the unfailing helpful
and constructive PV_Opt channel ;-)

That said, the latest update has been on the Solis-sensor integration.

I don’t really understand the whole version control / forking / merging
GitHub process. If I was choosing life again I’d have learnt to code
somewhere along the line !

Cheers,

Pete.

lazy-pete, 2025-08-03:
Seemed to deliver the plan overnight - discharged for a while in the evening then charged once the cheap rate kicked in. 

Does have a bunch of error logs at ~ 0200

[image]
[image]

[error.log](https://github.com/user-attachments/files/21565466/error.log)
[main.log](https://github.com/user-attachments/files/21565467/main.log)
[pv_opt.log](https://github.com/user-attachments/files/21565468/pv_opt.log)

stevebuk1, 2025-08-03:
I've completed the updates to Pv_opt to use Solis Sensor for the inverter status and Solis Control for the inverter control, this should end the need for any separate automations.  I've also put a fix in for the incorrect status reporting where it permanently said it was in a hold slot. I can't test tonight as I need an EV charge, but if its successful tomorrow I'll release a Beta for testing. 

I'll have a look at the issues with using Solis Sensors for both reads and writes from the inverter once the above is proved ultimately successful.

stevebuk1, 2025-08-04:
> Obviously happy to assist with testing as needed.

I'm introducing this as if its a seperate inverter integration, which I've called SOLIS_CLOUD_SENSOR_CONTROL. This maintains whats already in config.yaml for sorting out the control aspects of Solis Sensor seperately (which keeps its name of SOLIS_CLOUD). 

In prep for this can you send me a copy of your config.yaml,  I'll then make some edits and send it back to you. 

Please rename the file to config.yaml.log before tryiing to attach, otherwise Github will interpret it directly and include as text on a post, which is then unfortunately unrecoverable back into a file I can edit.

lazy-pete, 2025-08-05:
Hi Steve,

I think my config.yaml is attached correctly 

Cheers,

Pete

[config.yaml.log](https://github.com/user-attachments/files/21586214/config.yaml.log)

stevebuk1, 2025-08-05:
> I think my config.yaml is attached correctly

Thanks. Last nights testing didnt go well, I think I've got issues with threading/sequencing.

fboundy, 2025-08-05:
I haven’t been following this but I’ve just been asked to look at a bug in Solis Senor which may be contributing or confusing the issue: https://github.com/hultenvp/solis-sensor/issues/489#issuecomment-3152017619

stevebuk1, 2025-08-05:
> I think my config.yaml is attached correctly

Unfortunately its just a 17 byte file with the word "unauthorised" in it. 

I just checked mine and it attaches ok.

Try instead a copy and paste of the text but then select the whole lot and click the "code" button above (its the one that looks like <>). This is presuming you are on a PC/AppleMac.

stevebuk1, 2025-08-05:
I've solved the problems from yesterday and am about to release 5.0.0-Beta-3 for test. 

Appdaemon is flagging an occasisonal warning with regard to updating the times, I think thats because Solis Control handshakes with the inverter and sometimes it takes longer than the 10second AppDaemon limit for HA interaction. The logging will log this. It doesnt affect the write going through. 

Its not quite finished in that it currently writes to the inverter every optimisation run rather than only if it needs to modify the values already programmed, I'll sort that out this evening. 

1) Disable all additional automations and the SolaxModbus integration. 
2) Add the following to your config.yaml

```
 # ===============================================================================================================
  # Brand / Integration Specific Config: SOLIS_CLOUD_SENSOR_CONTROL
  # https://github.com/hultenvp/solis-sensor (for read only sensors) 
  # https://github.com/mkuthan/solis-cloud-control (for controlling the inverter)
  # ===============================================================================================================
  
  inverter_type: SOLIS_CLOUD_SENSOR_CONTROL
  device_name: solis
  update_cycle_seconds: 15
  
  # These are the default entities used with the Solis Sensor Integration (with Control Switched off) 
  # You can change them here and over-ride the defaults.
  
  # battery_voltage: sensor.{device_name}_battery_voltage_2
  # maximum_dod_percent: sensor.{device_name}_force_discharge_soc
  id_battery_current: sensor.solis_battery_current_2  # Debug purposes only
  # id_consumption_today: sensor.{device_name}_daily_grid_energy_used
  # # id_consumption:
  # #   - sensor.{device_name}_total_consumption_power
  # id_grid_import_today: sensor.{device_name}_daily_grid_energy_purchased
  # id_grid_export_today: sensor.{device_name}_daily_on_grid_energy
  # id_battery_soc: sensor.{device_name}_remaining_battery_capacity
  
  # These are the default entities used with the the Solis Cloud Control Integration.
  # You can change them here and over-ride the defaults.
    
  # id_inverter_mode: select.{device_name}_storage_mode

  # id_timed_charge_time: text.{device_name}_slot1_charge_time
  # id_timed_charge_current: number.{device_name_slot1_charge_current
  # id_timed_charge_soc: number.{device_name}_slot1_charge_soc

  # id_timed_discharge_time: text.{device_name}_slot1_discharge_time
  # id_timed_discharge_current: number.{device_name}_slot1_discharge_current
  # id_timed_discharge_soc: number.{device_name}_slot1_discharge_soc

  # Required for Tariff Comparison, see end of file. Can be commented out if not using. 
  #  id_solar_power:
  #  - sensor.{device_name}_dc_power_pv1
  #  - sensor.{device_name}_dc_power_pv2
```

3) To avoid mapping all the entities I'd recommend reinstalling Solis Control but when confirming the device name, delete the suggested text "inverter_control_XXXXXXXXXX" and replace with "solis". If not you'll need to map the Solis Control entities in config.yaml. You''ll need the storage switch, charge time, discharge time, charge current and discharge current. You can ignore target socs, these are for 6 slot firmware. 

4) I've added a new SolisCloud dashboard, mainly so the battery charge and discharge bars are still separate. Its here:
https://github.com/fboundy/pv_opt/blob/feature_soliscontrol/.dashboards/pvopt_dashboard_soliscloud_v2.yaml
The separate bars are generated by template entities, you'll need to add this to your configuration.yaml:
```

    - name: "Solis Battery Charge Power"
      unique_id: solis_battery_charge_power
      unit_of_measurement: W
      device_class: power
      state_class: measurement
      state: >-
        {{max(states('sensor.solis_battery_power_2') | float(0),0)}}    

    - name: "Solis Battery Discharge Power"
      unique_id: solis_battery_discharge_power
      unit_of_measurement: W
      device_class: power
      state_class: measurement
      state: >-
        {{max(-(states('sensor.solis_battery_power_2') | float(0)),0)}}

```

lazy-pete, 2025-08-05:
Great - will try and do that tomorrow.

Was copying file on iOS so maybe that’s why it didn’t work.

Cheers,

Pete

  1.  To avoid mapping all the entities I'd recommend reinstalling Solis Control but when confirming the device name, delete the suggested text "inverter_control_XXXXXXXXXX" and replace with "solis". If not you'll need to map the Solis Control entities in config.yaml. You''ll need the storage switch, charge time, discharge time, charge current and discharge current. You can ignore target socs, these are for 6 slot firmware.

  2.  I've added a new SolisCloud dashboard, mainly so the battery charge and discharge bars are still separate. Its here:
https://github.com/fboundy/pv_opt/blob/feature_soliscontrol/.dashboards/pvopt_dashboard_soliscloud_v2.yaml

stevebuk1, 2025-08-06:
> Its not quite finished in that it currently writes to the inverter every optimisation run rather than only if it needs to modify the values already programmed, I'll sort that out this evening.

I coded and tested this last night and it worked fine on IOG and an EV charge, preventing house battery discharge when the EV was charging. I'll roll this out at Beta-4, it would benefit from testing under a more dynamic profile that includes forced discharge (which I don't do as I have no export tariff). 

Will move onto looking at the control aspects of SolisSensor, at least as far as to see whether it works on my system or I have the same issues as you. This won't be till next week.

stevebuk1, 2025-08-06:
v5.0.0-Beta-4 is released that now supresses unnecessary inverter writes. 

Instructions for changing config.yaml are above but feel free to post config.yaml and I'll review/modify. 

Please run for 24 hours and post logs (including main.log/appdaemon.log), sooner if there are issues.

lazy-pete, 2025-08-06:
Thanks - sorry haven’t done it today. Got bogged down.

Will get to it as soon as I can and get some logs up.

Cheers,

Pete

lazy-pete, 2025-08-06:
So, apart from the new dashboard and templates to support it, I think I’ve got everything in, but not working. 

Disabled Solis cloud. Removed Solis-Control. Disabled my button press automation. 

Restarted HA. 

Added Solis-Control with name of ‘Solis’
Commented out all elements of previous Solis-cloud in config.yaml
Pasted in new Solis Control elements to yaml. 

Started Appdaemon. PV_Opt not happy. 

[Config.yaml.docx](https://github.com/user-attachments/files/21642476/Config.yaml.docx)

I was working on my phone so word was the best I could manage - can try again on desktop later if this doesn’t work. 

[main.log](https://github.com/user-attachments/files/21642486/main.log)
[pv_opt.log](https://github.com/user-attachments/files/21642485/pv_opt.log)
[error.log](https://github.com/user-attachments/files/21642484/error.log)

lazy-pete, 2025-08-06:
Might have found the problem - despite changing the name at setup the integration has named all the entities as inverter_control_xxxxxxxxxxxxxxxx 

Rather than rename everything I’ve just changed the name in the yaml to that above and rebooted appdaemon.

stevebuk1, 2025-08-07:
Looks like it can't find `sensor.solis_battery_current_2`. My system has entity `sensor.solis_battery_current_2` associated with solis-sensor, as the SolaxModbus integration previously generated one called `sensor.solis_battery_current`. Depending on how long the SolaxModbus integration was disabled for, the entities might have been removed the last time you installed solis-sensor, so it might be called `sensor.solis_battery_current`. Try that name in config.yaml. 

sensor.solis_battery_voltage_2 was the only other overlap in SolaxModbus and Solis-Sensor but I don't think thats used by Pv_opt.

lazy-pete, 2025-08-07:
> You've also set device_name within config.yaml to
> inverter_control_603102322a140096.
>
> Whilst this then means you don't need to manually override the entity
> mapping in config.yaml for solis-control, it means you will then need to
> overide the mapping for solis-sensor, which uses solis as the device
> name. This is why I installed solis-control using the device name of solis,
> so the single value of device_name as set in PV_opt is suitable for both
> integrations without any mapping. (I did toy with introducing a 2nd device
> name setting called device_name_2 or something, but it was getting way too
> complex)
>
> Anyway, once you've sorted sensor.solis_battery_current I think you'll
> need to override the names of the rest of the solis-sensor entities so
> Pv_opt can find them.

Thanks. Will have a fiddle.

I thought I no longer needed Solis Sensor so the name didn’t matter ?

Cheers,

P

lazy-pete, 2025-08-07:
Understood. 

I reinstalled Solis-control and this time it labelled entities properly as Solis. 

Few yaml tweaks (I don’t have any _2 versions) and seems to be up and running. Will now try disabling the control element of Solis-sensor. 

[main.log](https://github.com/user-attachments/files/21673212/main.log)
[error.log](https://github.com/user-attachments/files/21673214/error.log)
[pv_opt.log](https://github.com/user-attachments/files/21673213/pv_opt.log)

[# Internal configuration --- DO NOT.txt](https://github.com/user-attachments/files/21673223/Internal.configuration.---.DO.NOT.txt)

stevebuk1, 2025-08-07:
I think it halted at 21:47 still looking for  sensor.solis_battery_current_2?

I can see sensor.solis_battery_current is reporting valid values so change your config.yaml from:

`# id_battery_current: sensor.solis_battery_current # Debug purposes only`
to
`id_battery_current: sensor.solis_battery_current # Debug purposes only`

i.e delete the # from the start of the line

lazy-pete, 2025-08-07:
Thanks - worked that out eventually from the error logs. 

Got most of the dashboard working too after a bit of fiddling, except house load today / yesterday, but I think those may just need time to populate as it’s been wonky for a couple of days. 

Seems to be running now ….

[# Internal configuration --- DO NOT.txt](https://github.com/user-attachments/files/21673934/Internal.configuration.---.DO.NOT.txt)

[main.log](https://github.com/user-attachments/files/21673943/main.log)
[pv_opt.log](https://github.com/user-attachments/files/21673942/pv_opt.log)
[error.log](https://github.com/user-attachments/files/21673941/error.log)

lazy-pete, 2025-08-07:
Cool. Something isn’t quite right - it looks as if it should be discharging, however my sensor.solis_battery_power is stubbornly 0 (had to fiddle a bit with templates for the dashboard, however reports 0 from the integration). 

More logs tomorrow night.

stevebuk1, 2025-08-08:
> however my sensor.solis_battery_power is stubbornly 0 (had to fiddle a bit with templates for the dashboard, however reports 0 from the integration).

Just checked the logs (Pv_opt records the value of all entities with the word "solis" in them) and agree its always zero. I don't think it affects Pv_opt operation, just your dashboard won't give you battery power. I note the voltage and current are corrrect:

```
23:46:35     INFO:   sensor.solis_battery_voltage (voltage):                                     53.5 V
23:46:35     INFO:   sensor.solis_battery_current (current):                                     -0.7 A
```
 so you could just multiply these together to get power and then display that.

lazy-pete, 2025-08-08:
Thanks. I can work on the last of the dashboard. 

Meantime I think something is still not right. 

More logs. 

[main.log](https://github.com/user-attachments/files/21692400/main.log)
[error.log](https://github.com/user-attachments/files/21692401/error.log)
[pv_opt.log](https://github.com/user-attachments/files/21692399/pv_opt.log)

lazy-pete, 2025-08-09:
Think I’ve perhaps got it sorted. 

Realised belatedly that following the wild weather here start of the week my inverter grid connection had tripped resulting in a ‘No-grid’ error that I hadn’t got / noticed. 

So whilst PV-opt was writing to the inverter it was incapable of delivering PV / battery charge / discharge to the house / grid. D’oh. 

A bit more fiddling has got almost all the dashboard working as well, although I am curious as to what entity you are using for sensor.solis_house_load_today and sensor.solis_house_load_yesterday ? I can’t find those entities and I’d have thought they’d exist, even if the state was inaccurate for to the grid issue. 

Had to swap around states and a couple of entity names (_2) of the helpers to get the first bit all going. Watching it to see if I’ve got them all the right way around. 

[error.log](https://github.com/user-attachments/files/21701274/error.log)
[main.log](https://github.com/user-attachments/files/21701273/main.log)
[pv_opt.log](https://github.com/user-attachments/files/21701275/pv_opt.log)

[# Internal configuration --- DO NOT (1).txt](https://github.com/user-attachments/files/21701316/Internal.configuration.---.DO.NOT.1.txt)

stevebuk1, 2025-08-09:
> A bit more fiddling has got almost all the dashboard working as well, although I am curious as to what entity you are using for sensor.solis_house_load_today and sensor.solis_house_load_yesterday ? [...]

I'm using sensor.solis_daily_grid_energy_used. Its an incrementing kWh sensor but I have noticed its decrementing as well (should never do that!), unsure of what effect that will have on half hour consumptions:

[image]

stevebuk1, 2025-08-09:
> Think I’ve perhaps got it sorted.
> 
> Realised belatedly that following the wild weather here start of the week my inverter grid connection had tripped resulting in a ‘No-grid’ error that I hadn’t got / noticed.

I've stared at your logs and can conclude that everythings been written to the inverter as it should, it was just you weren't getting charging/discharging.  So sorta relieved you found another issue, otherwise we'd pretty much reached the end of the road!

stevebuk1, 2025-08-09:
For info, I've just removed solis-control and enabled the control portion of solis-sensor, and am getting the original issue you first reported i.e the energy control switch always reports unknown. It does actually work - I can set it on one Rpi and get the value picked up on another Rpi (still running solis-control) and also the Solis App, it just doesn't report a value (or any of the possible options if you try and read these into Appdaemon). I've put a fix into Pv_opt so it deals with Unknown and just carries on.

stevebuk1, 2025-08-10:
> I’ll leave it running via Solis Control for now, although I’ve still got the Solis Sensor Control entities visible anyway.

Please do, as I haven't finished testing the fix yet and its not yet released.

fboundy, 2025-08-13:
I've just submitted a PR for Solis Sensor which fixes two bugs in it: https://github.com/hultenvp/solis-sensor/pull/493

stevebuk1, 2025-08-13:
> I've just submitted a PR for Solis Sensor which fixes two bugs in it: [hultenvp/solis-sensor#493](https://github.com/hultenvp/solis-sensor/pull/493)

Out of interest does the Energy Storage control switch on your system show as Unknown? (issue #443 on Solis Sensor)

[image]

stevebuk1, 2025-08-14:
> > No but the bug only affects certain firmware versions. The six-slot firmware came in at v 4B and someone had mis-coded it as 42 so if you have 43-4B it was giving you the wrong setup. Mine is 4C so it was fine anyway
> 
> Ok, your PR doesnt affect Pete or I as I have 3D and he has 42, both of us have the storage control switch reporting as Unknown. I'm guessing you had your 6 slot firmware upgrade in place before coding the control part of Solis Sensor, so its always worked for you.
> 
> I'll see if I can work out whats causing the issue. The writes definitely work. If i remember rightly there are seperate modbus registers for read and write for the energy storage control switch, perhaps the read register moved between the 3 slot and 6 slot firmware, or more correctly the API call doesnt remain the same.

fboundy, 2025-08-14:
If you turn on debug logging for Solis Sensor I can try to have a look at what’s failing

stevebuk1, 2025-08-14:
> If you turn on debug logging for Solis Sensor I can try to have a look at what’s failing

Turned on logging, 35d is coming back ok but the lookup isnt returning a select option:

```
2025-08-14 20:54:49.514 DEBUG (MainThread) [custom_components.solis.select] >>> Update for Solis Energy Storage Control Switch
2025-08-14 20:54:49.514 DEBUG (MainThread) [custom_components.solis.select] >>> Current option: None
2025-08-14 20:54:49.514 DEBUG (MainThread) [custom_components.solis.select] >>> Value: 35
2025-08-14 20:54:49.514 DEBUG (MainThread) [custom_components.solis.select] >>> Lookup: None
2025-08-14 20:54:49.514 DEBUG (MainThread) [custom_components.solis.select] >>> Last updated: 2025-08-14 20:54:49.492259
```
Its because the ALL_CONTROLS False (3 slot option) lists this for the energy storage control switch:

```
        "636": [
            SolisSelectEntityDescription(
                name="Energy Storage Control Switch",
                key="energy_storage_control_switch",
                option_dict={
                    1: "Self-Use - No Grid Charging",
                    3: "Timed Charge/Discharge - No Grid Charging",
                    17: "Backup/Reserve - No Grid Charging",
                    33: "Self-Use - No Timed Charge/Discharge",
                    35: "Self-Use",
                    37: "Off-Grid Mode",
                    41: "Battery Awaken",
                    43: "Battery Awaken + Timed Charge/Discharge",
                    49: "Backup/Reserve - No Timed Charge/Discharge",
                    51: "Backup/Reserve",
                    64: "Feed-in priority - No Grid Charging",
                    96: "Feed-in priority - No Timed Charge/Discharge",
                    98: "Feed-in priority",
                },
                icon="mdi:dip-switch",
```
 wheras the 6 slot option lists this:

```
        "636": [
            SolisSelectEntityDescription(
                name="Energy Storage Control Switch",
                key="energy_storage_control_switch",
                option_dict={
                    "1": "Self-Use - No Grid Charging",
                    "3": "Timed Charge/Discharge - No Grid Charging",
                    "17": "Backup/Reserve - No Grid Charging",
                    "33": "Self-Use - No Timed Charge/Discharge",
                    "35": "Self-Use",
                    "37": "Off-Grid Mode",
                    "41": "Battery Awaken",
                    "43": "Battery Awaken + Timed Charge/Discharge",
                    "49": "Backup/Reserve - No Timed Charge/Discharge",
                    "51": "Backup/Reserve",
                    "64": "Feed-in priority - No Grid Charging",
                    "96": "Feed-in priority - No Timed Charge/Discharge",
                    "98": "Feed-in priority",
                },
                icon="mdi:dip-switch",
```
i.e. the key value is in quotes. Adding quotes to the values has solved the issue on my system. 

I'll update #443 on SolisControl with this info.

stevebuk1, 2025-08-15:
> Tag me on it and I’ll raise a PR

Tagged

lazy-pete, 2025-08-20:
> Apologies for having g you hunt through my logs for a fix when I haven’t
> had a grid connection since the 4th ! I thought the Solis app sent me
> notifications if that happened - shall have to check settings. I’ll leave
> it running via Solis Control for now, although I’ve still got the Solis
> Sensor Control entities visible anyway.
>
> Hi @lazy-pete <https://github.com/lazy-pete>, hows this working for you
> now? I've swapped my main system back to modbus as the issue with the house
> consumption reporting on SolisCloud doesnt look like its going to get fixed
> anytime soon. If you are up and running with Solis Sensor and Solis Control
> in tandem then I'll close this issue.

Hi Steve,

It seems pretty stable, in that I’ve ignored it for a couple of weeks and
it seems to do its thing. Gets hung on updating inverter / read cycles
occasionally but seems to sort itself out after a while.

Been watching the progress on the integration PR’s from yourselves and just
thought I’d wait version upgrades before trying to update anything. I
presume the fixes you’ve made thus far will make it into the next stable
PV-opt release ? I’m still on the beta. Same question for the Solis cloud /
control ones ?

Cheers,

P

stevebuk1, 2025-08-21:
> Been watching the progress on the integration PR’s from yourselves and just thought I’d wait version upgrades before trying to update anything. I presume the fixes you’ve made thus far will make it into the next stable PV-opt release ?

If its working ok then I'll do a PR to get Pv_opt released that includes the configuration option to use Solis Sensor and Solis Cloud together. Theres a few tidyups to do e.g. removing the extra battery current monitoring I added, will get to it.

Steve

lazy-pete, 2025-08-22:
Great.

solarjones, 2025-09-01:
Interestingly, I'd loaded Solis Cloud integration (more out of curiosity) but hadn't commissioned it, and it appeared to break PV_Opt for me too.

[error.log](https://github.com/user-attachments/files/22082091/error.log)
[main.log](https://github.com/user-attachments/files/22082092/main.log)
[pv_opt.log](https://github.com/user-attachments/files/22082093/pv_opt.log)

Uninstalling the Solis integration resolved it for me.  My inverter firmware version is 4A004C if that helps.