stevebuk1/pv_opt #414: Unexpected error running initialize() for pv_opt

solarjones, 2025-09-20:
Hi,
Need a pointer on what I've done wrong. Trying to get the Car integration element working and have replaced the config.yaml as recommended together with the dashboard.  However, I can't seem to get things working.

Whenever I select Enable Agile Car Charge Control (Zappi already selected), it doesn't stay selected.

[image]

Also receive the startup error.

[pv_opt.log](https://github.com/user-attachments/files/22446977/pv_opt.log)
[error.log](https://github.com/user-attachments/files/22446976/error.log)
[main.log](https://github.com/user-attachments/files/22446978/main.log)

Sorry if I'm being a bit stupid, but can't figure out what to do as the car options just aren't displayed currently.

Thanks in advance!
Simon

stevebuk1, 2025-09-20:
The error being generated is related to modbus setup. Are you running the Solax Modbus integration to your inverter?

If so, can you post a screenshot of whats in the dropdown menu when you expand "Energy Storage Control switch":

[image]

if not, please post your original config.yaml.

stevebuk1, 2025-09-21:
> I am using the SolaX integration.
> image.png (view on web)
> <https://github.com/user-attachments/assets/fbe56915-887c-4057-a6be-f0928d3f091e> image.png
> (view on web)
>
> Thanks

Are you still on appdaemon 0.16.7?

solarjones, 2025-09-21:
I am. Should I update and use the latest beta of pvopt?Sent from my iPhone

stevebuk1, 2025-09-21:
No, not for the moment, as Pvopt 4.0.11 is only suitable for 0.16.7. I
don't think I updated the config.yaml on the repo for anything beyond
4.0.11 but will recreate this reconfiguration to see if I can create the
same error.

stevebuk1, 2025-09-21:
Hmmm. On my development system, when I disabled SolisCloud, renenabled Modbus and restarted Pv_opt, I got exactly the same error as you did. I thought it was to do with the 3 slot/6 slot firmware switch but mine was in the correct position (I asssume you have 6 slots?). Restarting Pv_opt then cleared it. 

So, I'd restart Pv_opt (just add a carriage return or other some edit to config.yaml) and see if that clears it, if not, restart Appdaemon.

If you are on 3 slot firmware, change `hmi_firmware_fb00_plus: True` to `hmi_firmware_fb00_plus: False` in config.yaml.

stevebuk1, 2025-09-22:
Regarding EV battery capacity, with `overwrite_ha_on_restart: false` in config.yaml then when Pv_opt initialises it will pick up the value from the slider and not change it to whatever is set in config.yaml. If you always want the values in config.yaml to be used on a Pv_opt restart, then change `overwrite_ha_on_restart: false` to true.  

However, as implied above, the slider values for EV Charger Power and EV Battery capacity are only read on restart. If you change the slider values then the new ones will not be used for the next optimiser run. I figured these values never change so they are best set in config.yaml and forgotten about, but appreciate that if the sliders are changed then Pv_opt does an optimisation run, which implies the new values are used. I'll change this to be consistent with the other (non-EV) sliders like house battery capacity so if the optimiser is triggered by a change of slider, the new value is used.

Regarding Enable Agile Car Control, I can't repeat your behaviour. The dashboard setting is always respected and stays there. 
If I change `control_car_charging` in config.yaml with `overwrite_ha_on_restart: false` then it has no effect on the dashboard value. 
If I change `control_car_charging` in config.yaml with `overwrite_ha_on_restart: true` then the value of `control_car_charging` updates the dashboard value. 

Can you post a Pv_opt.log that includes a restart (just change a value within config.yaml), and then a dashboard toggle of Enable Agile Car Control?

stevebuk1, 2025-09-22:
> Thanks for investigating. My inverter has 3x charge/discharge slots and I did have a firmware update last month. Running 4A004C-050000. Didn't know there was a 6 slot option on RHI-6K-48ES-5G.

I wasnt sure what was available on what. I also have a 48ES-5G, but my firmware has been unchanged for several years (something I should probably get resolved). With 3 slot firmware, `hmi_firmware_fb00_plus: False` is correct.

stevebuk1, 2025-09-28:
No problem. I'm on IOG so can better assist with any unexpected behaviors.  Not so long ago IOG generated a schedule when you plugged your car in and it never changed for the night, but now things are very dynamic. I've made some mods to Pv_opt to better cope with this but it there still can be a 10 minute delay between Octopus reschedules and Pv_opt picking it up. This is most apparent if Pv_opt is being used to prevent house battery discharge (which is my use case).

stevebuk1, 2025-09-28:
Ok good luck! Makes sense to ensure everythings working before bringing Pv_opt into play. 

 If you've got a Zappi then its best that IOG controls that rather than the car (one or the other are possible, depending on car make). You'll know which via the Octopus App, if its Zappi then the devices tab will say "Charge to add" rather than "Charge Target". Either will work fine with Pv_opt, I had the car integrated before Octopus released the Zappi integration, then I swapped.